arcao/Syslog #3: add ability to specify appName on a per msg

sticilface, 2017-02-01:
Hi,

use senario:  I have several libraries running on the same ESP, doing a variety of things. With debugging enabled I'd like to be able to send debug messages from different bit of software on the same device.  so the device name is the same, but the app name is different.  

Would be great to have overloads to be able to specify the log level and app name for a message. 

what do you think?

arcao, 2017-02-01:
Yes it's possible to implement. There is only one problem with `logf` functions without `pri` parameter because of varargs. You cannot have defined both `logf(const char *, ...)` and `logf(const char *, const char *, ...)`.

Anyway in one of my project I call `appName` before `log` / `logf` methods to do same as you describe:

```c++
IPAddress ip = WiFi.localIP();
syslog.appName("network").logf(LOG_INFO, "Assigned %s.local [%u.%u.%u.%u]", DEVICE_HOSTNAME, ip[0], ip[1], ip[2], ip[3]);
```

Only caveats is that `appName` store the value for following calls of `log` / `logf`. So you have it everywhere.

sticilface, 2017-02-02:
yes. i realised that.  and there is no function to fetch it, set the new one, then set it back to the old one!

sticilface, 2017-02-02:
i wonder what the most elegant solution would be.  Think its worth having!

sticilface, 2017-02-22:
I've come up with a method that might work.  I'm going to try an roll it into my own syslog attempt for my ESPManager software.  I've written a string class with derived versions `myStringf` and `mystgingf_P` with constructors that take the `...`.  This allows you to do this:
```
log( "appName", myStringf_P( PSTR("Test %u"), "123" ); 
log( "appName", myStringf("Test %u", "123" ); 
log( "appName", "123" ); 
```
Using the right move semantics makes it efficient...

sticilface, 2017-02-23:
I've use some of your code to write my own syslog client, mainly the log mask stuff and message generation.  I've given you credit for it at the top of the `.h` is that ok.  There is no licence in the repo so I want to check? Its all none profit, open source stuff.

sticilface, 2017-03-07:
I closed this by accident.  sorry.