arcao/Syslog #4: vsnprintf_P to vsnprintf

sticilface, 2017-02-14:
This took a long time for me to track down.   Using the Arduino IDE for ESP8266. 

In a nutshell, including syslog causes `vsnprintf_P` to be redefined as `vsnprintf`. 

Tested by including a #pragma message("redefined") within the if !defined block. 
```cpp
#if !defined(ARDUINO_ARCH_AVR) || !defined(ARDUINO_ARCH_ESP8266) || !defined(vsnprintf_P) || !defined(ESP8266)
 #pragma message("vsnprintf((buf), (len), (fmt), (args))")
 #define vsnprintf_P(buf, len, fmt, args) vsnprintf((buf), (len), (fmt), (args))
#endif
```

output from compiler
```
In file included from /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.h:59:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp:2:
/Users/amelvin/Documents/Arduino/libraries/Syslog/src/Syslog.h:19:58: note: #pragma message: vsnprintf((buf), (len), (fmt), (args))
  #pragma message("vsnprintf((buf), (len), (fmt), (args))")
```

The result of this is that use of it in that lib causes an immediate crash as it using the standard vsnprintf function.  Removing the ` #define vsnprintf_P(buf, len, fmt, args) vsnprintf((buf), (len), (fmt), (args))` causes everything to work fine.  

This took a lot of weirdness to figure out, especially as vsnprintf_P functions for serial were still working.   The odd thing is that these defines do exist.. `ARDUINO_ARCH_ESP8266` and `ESP8266`, which i added. 

any ideas?

sticilface, 2017-02-14:
changing it to this works
```
#if !defined(ARDUINO_ARCH_AVR)
#if !defined(ARDUINO_ARCH_ESP8266)
#if !defined(vsnprintf_P) 
#if !defined(ESP8266)
 #pragma message("vsnprintf((buf), (len), (fmt), (args))")
 #define vsnprintf_P(buf, len, fmt, args) vsnprintf((buf), (len), (fmt), (args))
#endif
#endif
#endif
#endif
```

sticilface, 2017-02-14:
I guess actually... the logic needs `&&` not `||`
```cpp
#if !defined(ARDUINO_ARCH_AVR) && !defined(ARDUINO_ARCH_ESP8266) && !defined(vsnprintf_P) && !defined(ESP8266)
 #pragma message("vsnprintf((buf), (len), (fmt), (args))")
 #define vsnprintf_P(buf, len, fmt, args) vsnprintf((buf), (len), (fmt), (args))
#endif
```

For the condition to be met you want them all to be true, if one is present then do not define

arcao, 2017-02-14:
Fixed in 532370ce893ab7f762a7f5e083c3af34ef0e7cc6.

closed 2017-02-14 by arcao