arcao/Syslog #5: Allow multiple outputs

sle118, 2017-02-22:
@arcao 

I would like to suggest a feature for this project. 

Given that it might be interesting to be able retrieve logs that might have been created while a device didn't have access to the network or while the syslog server was down, it would be nice to offer an option to have a buffer page of a predetermined size and periodically rotate it. In order to minimize the impact to the memory, the logs could be written to a SPIFFS file.  Maybe this is an overkill for the flash memory, I don't know what sort of abuse it can sustain in terms of write/erase cycles.

Please let me know your thoughts.

arcao, 2017-04-17:
👍  I like this idea. More **appenders** like `UdpSyslogAppender`, `BsdUdpSyslogAppender`, `PrinterSyslogApender`, `FileSyslogAppender`, etc.

and usage like:
```cpp

// define
UdpSyslogAppender udpAppender(udpClient, SYSLOG_SERVER, SYSLOG_PORT, DEVICE_HOSTNAME);
PrinterSyslogApender serialAppender(Serial);
Syslog syslog(APP_NAME, LOG_KERN, udpAppender, serialAppender);

// and use
syslog.log(LOG_INFO, "This is info log message");
```

sle118, 2017-04-19:
Would it be an Overkill to actually have the appenders register with delegates? This would allow runtime flexibility of adding and removing appenders. 

Aside from this, thanks for including it as a milestone!

kluzzebass, 2017-11-24:
What you guys are very close to describing is a generic logging service (like log4j), in which syslog is merely one of many appenders, along with Serial, File, etc.

Several of these exist (albeit in various embryonic stages), but they all seem to fall short of actually being flexible enough to also handle syslogging.

Perhaps version 3 of this library will be the logging framework we've all been looking for?