arcao/Syslog #6: Disconnected operations

MarcFinns, 2017-03-02:
Hi Arcao,
I have a project that needs to run also when disconnected from Wifi.
I integrated your lib and have a number of syslog statements in my code. What happens when syslog is called and connection is not there? is it handled gracefully or is it an error condition? Does it lock until a timeout? please let me know.
thanks!

MarcFinns, 2017-03-02:
Specifically, if the lib does not handle it already I propose to add a setConnected(bool) API and check if(connected) before doing network stuff. 
Please let me know if you would accept a pull request... thanks

arcao, 2017-03-07:
At the moment it's handled gracefully. In a next major version `log` and `logf` if message was sent successfully (see note bellow). 

Note: This library use UDP protocol which is stateless, thus there is no way to check if server received log message or not.

MarcFinns, 2017-03-07:
Thanks for your time. It is clear that with udp you can't check it messages are received... I just wanted to make sure that the call does not crash if disconnected. That's good.  Thanks again!!

arcao, 2017-04-17:
This is fixed in #3.

closed 2017-04-17 by arcao