arcao/Syslog #19: Using syslog.logf crashed the Arduino

KarlLovink, 2018-10-07:
I'am using the libraty on a Arduino Mega with an Ethernet Shield. When I use the syslog.logf function the Arduino stops working.  Any idea what I' am doing wrong.

Sketch:

[RemotePowerSwitch.txt](https://github.com/arcao/Syslog/files/2454248/RemotePowerSwitch.txt)

saspol, 2019-04-02:
Have same problem with Mega and W5100 shield.
First usage syslog.log() crashing system.

brianrho, 2020-09-27:
Likely caused because the library doesn't open the Ethernet socket first before writing log strings to it. This should be done with `this->_client->begin(someRandomLocalPort)` or in your sketch `setup()`, can also be done with `udpClient.begin(someRandomLocalPort)`. 

At first, while trying to port this library, I spent a while searching to see if this was already done automatically somewhere down the stack, but I found nothing. Then I found this issue which confirms the present Ethernet example won't work, probably wasn't a big deal before since most people use ESP8266/32 WiFi, whose UDP library doesn't seem to directly operate on sockets the same way as the W5x00 library, instead dynamically creating a new UDP context as needed for each datagram.

arcao, 2026-10-12:
Fixed in [v2.0.1](https://github.com/arcao/Syslog/releases/tag/v2.0.1). The crash had two causes: the Ethernet examples did not open the UDP socket (`udpClient.begin(localPort)` is required with the W5x00 Ethernet library, see README → Usage notes), and `logf` reused a `va_list` for longer messages (#29). Please reopen if you still see the problem with v2.0.1.

closed 2026-10-12 by arcao