arcao/Syslog #20: Log Level

mich4el-git, 2018-10-14:
Hello,
I have question to the Level of the Logs. I successfully send the logs to a raspberry which is recording the logs using syslog-ng.
However I don't see the log level in the log file on the raspberry.
Here the result from the ESP8266SimpleLogging.ino

`Oct 14 18:59:48 192.168.0.101 1 - my-device my-app - - - ﻿This is info message no. 1
Oct 14 18:59:48 192.168.0.101 1 - my-device my-app - - - ﻿This is daemon info message no. 1
Oct 14 18:59:48 192.168.0.101 1 - my-device my-app - - - ﻿End loop
Oct 14 18:59:58 192.168.0.101 1 - my-device my-app - - - ﻿Begin loop
Oct 14 18:59:58 192.168.0.101 1 - my-device my-app - - - ﻿This is error message no. 2
Oct 14 18:59:58 192.168.0.101 1 - my-device my-app - - - ﻿This is info message no. 2
Oct 14 18:59:58 192.168.0.101 1 - my-device my-app - - - ﻿This is daemon info message no. 2
Oct 14 18:59:58 192.168.0.101 1 - my-device my-app - - - ﻿End loop
`
Is there a possibility to see the log level using syslog-ng?

tablatronix, 2021-02-07:
Yeah doesn't work for me, 

`Feb  6 22:39:38 Host App ﻿Message`

Should be ``Feb  6 22:39:38 Host PRI: App ﻿Message`

arcao, 2026-10-12:
The level is sent — it's encoded in the PRI part of the packet (`<pri>` at the very beginning). The problem is on the syslog-ng side:

1. The library sends messages in IETF format (RFC 5424) by default. Your output (`1 - my-device my-app - - - ...`) shows that syslog-ng parsed it as the old BSD format and put the RFC 5424 header into the message text. Use the `syslog()` source driver (or `flags(syslog-protocol)` on `network()`), which understands RFC 5424:

   ```
   source s_esp {
     syslog(transport("udp") port(514));
   };
   ```

2. The default file template of syslog-ng does not contain the level at all, you have to add it with a custom template, e.g.:

   ```
   destination d_esp {
     file("/var/log/esp.log" template("${ISODATE} ${HOST} ${PROGRAM} ${FACILITY}.${LEVEL}: ${MSGHDR}${MSG}\n"));
   };
   ```

Alternatively you can switch the library to BSD format (RFC 3164) by passing `SYSLOG_PROTO_BSD` as the last argument of the `Syslog` constructor, which older syslog servers parse out of the box.