arcao/Syslog #24: Is the library still maintained?

Rocco83, 2021-05-14:
There are several issues opened since years...
Is this library maintained?
If not, is a fork maintained?
Or is there any alternative?

mraction, 2025-02-25:
Thats a good question...

arcao, 2026-10-12:
Sorry for the long silence. Yes, the library is still maintained. I went through the open issues and #29 fixes the long-standing bugs:

- crash / garbage output in `logf` and `logf_P` for longer messages (#18),
- examples for Ethernet shield / WiFi shield not opening the UDP socket, which caused nothing to be sent or the board to crash (#19, #26).

The remaining questions are answered in their issues.

arcao, 2026-10-12:
[v2.0.1](https://github.com/arcao/Syslog/releases/tag/v2.0.1) is out with the bug fixes, new WiFiNINA and ESP32 examples and CI on GitHub Actions. Closing, feel free to open new issues.

closed 2026-10-12 by arcao