arcao/Syslog #26: Pass IP address instead of server name

tuctboh, 2022-08-10:
Hi,

I run a home lab and want to specify the IP address of the SYSLOG_SERVER, but seems like it only is looking for a string. (Since I'm tcpdumping the destination and nothing is coming to 514).

Tnx, Tuc

tuctboh, 2022-08-12:
So dug into this more..

1) Learned I could put #define SYSLOG_SERVER {192,168,3,245} and it actually worked just fine
2) Realized after doing some investigation about WiFiNINA and UDP, you need to issue a client.begin(localPort). 

I've submitted pr #27 to add an example file for WiFiNINA for others.

DrGonzo65, 2023-09-04:
Thank you! You saved me on this one. Needed to add the client.begin(localPort)

arcao, 2026-10-12:
Fixed by #29 (examples now call `udpClient.begin()`) and #27 (new WiFiNINA example). With WiFiNINA / Ethernet shield the UDP socket has to be opened with `udpClient.begin(localPort)` before logging. An IP address can also be passed directly instead of the server name, e.g. `IPAddress(192, 168, 1, 10)`.

closed 2026-10-12 by arcao